ktand/Armed #2: Kicad schematic shows errors when opening

xebbmw, 2019-07-16:
Hi, 

When I am trying to open the schematic file using Kicad 5, there are error messages showing libraries missing:
Armed-rescue.lib
object-technologies.lib

Could you please upload the above libraries to Github?

zedthirteen, 2020-03-25:
I am also having this issue with missing libraries

ktand, 2020-12-21:
I will add the missing files in the next few days!

zedthirteen, 2020-12-21:
> 	Virus-free. www.avg.com

Really nice to hear from you Karl.

I love your design. Thanks for all of your efforts.I certainly will appreciate the missing files. I was able to map many of the components to standard library items but not the FETs or the STM32 :-(

Should my i2c voltage be 3.3 or 5v?  My PC is in for repair just now, so not sure where I got to,  but I think I tried 3.3v and the writes failed - might be due to some other error though

Thanks again
Regards
David

ktand, 2020-12-21:
Sorry for missing all the notifications. They got buried in all other
github-notifications. I'll try to be more responsive!

I'm running the EEPROM @ 3.3V - with everything working.

zedthirteen, 2021-01-22:
Hi Karl, sorry to nag - but any news on those missing libraries? Are you having to recreate them?

one of them was in  C:/Users/Karl/PCB-designs/KiCad/Libraries/object-technologies.lib on your PC. The armed-rescue.lib file has environment variables in its embedded file name so I can't help with that one.  I think the object-technologies library had the missing STM32 component.   I think armed-rescue had many missing items including BSH105 and IPD036N04L.

I managed to find suitable standard library or alternative symbols for all except the STM32F407, BSH105 and IPD036N04L

ktand, 2021-01-22:
Sorry, I have updated the schematic to use the armed.lib library (commit c9e9fdd157b36a4a59312abd6035c331b24d473f). The armed.lib now contains the referenced symbols from object-technologies.lib. Please let me know if it works for you.

zedthirteen, 2021-01-22:
looks perfect now. Opens without errors and I can now see which processor pins connect to which signals - thanks :-)

closed 2021-01-22 by ktand