ktand/Armed #5: Support STM32 Armed

BorPass, 2020-11-29:
Dear Karl, dear community,
I want to replace my old gt2560 printer board with this STM32 Armed board.
I managed to compile Marlin 2.0 for the board with arduino ide and uploaded with STM32CubeProgrammer.
The problem I am currently facing and for which can‘t find a solution is, that I can‘t get access to the board through usb-port using octoprint, which is my print-server.

What I did so far:

1. I uploaded a blink sketch the same way I uploaded Marlin, in order to check my upload process. It worked, the heartbeat led is blinking!
2. Marlin compiles without errors. Therefore I guess the hex-file it creates is correct. By the way, I used Serial_Port -1 in configuration.h, which was the only configuration that compiled without errors.
3. dmesg shows the usb-port of my armed board as /dev/ttyACM0

Do you have any ideas, what else I could check in order to find the problem or which files I should share to give you a hint to the problem.

Another question:
Is there a description of how to connect a lcd display to the board.

Best Regards

ktand, 2020-12-30:
Did you configure Marlin to use the USB serial port?

BorPass, 2020-12-30:
Hello Karl, thanks for the reply. I discussed my issue under the subject of David (zedthirteen), with whom you had contact in the last days.
My current status is:
- usb works
- Simulated eeprom works, real eeprom not, because the chip is missing on my board. I‘m pretty sure real eeprom will work, as soon as I soldered the already ordered chip.
- I still have problems with the watchdog- part of Marlin. It causes my board to restart every few seconds - commenting it out, it works. I will check your 2.0.7 version of Marlin tomorrow for differences to my current version. 
- More work will have to be done for me on using tmc2208 in uart mode. Should this work and if yes, do you have any advice if I have to solder bridges like David did for making his tmc-stepper work in spi mode?

I think I‘m on the way to success...

It would be perfekt, if you publish some kind of manual for several details, e.g the ones I found in the pins_ARMED.h concerning the rx/tx pins for stepper-communication. It would be useful to have this in the board documentation here, perhaps together with a hint, why rx and tx is the same pin. As far as I thought to know, you need them separate, but I did not read enough about that. Information on the purpose of the several soldering bridge possibilities would be great, too, especially, because they are very board-specific. 

Anyway, the more I understand about your board, the more I like it.

But it is still some way to go for me, until it will work like I want it to do.

Best Regards,
Boris

ktand, 2021-01-22:
Hi, sorry, no manual is planned at this time.

Maybe you can get some help from the configuration I'm using. (TMC2209), please take look at my Marlin branch here https://github.com/ktand/Marlin/tree/2.0.7.2-armed-12864-2020-11-05

closed 2021-01-22 by ktand