ktand/Armed #6: Fan control

zedthirteen, 2021-01-07:
Does fan control work? 

Or, am I missing something in the way I am connecting the fan or some extra required configuration perhaps?

My TMC2130's are overheating so I added a 24V fan to the electronics door, connected to FAN0 (also tried both FAN1 and FAN2) but I was unable to get it to work. My board has 24V supply.

I tried sending fan control g-code commands (M106 P0 S255 and other variations) but was unable to get the fan to activate.  I have since hardwired the fan to 24V supply but would rather have control over it if possible.

ktand, 2021-04-15:
Fan control works. Make sure you build Marlin using the latest Platform IO. There was a problem with the board definition in an earlier release of PlatformIO / Arduino Core STM32 that caused a fan control issue.

zedthirteen, 2021-04-15:
Thanks for feedback. I will try as you suggest

zedthirteen, 2021-06-24:
Finally found some time to experiment with builds.
Still having a couple of problems - using your fork (2.0.8.2-armed-12864-2021-06-01) and my configuration(adv).h files. To the best of my knowledge PlatformIO is up to date, I always agree to update and restart. Core is 5.1.1, Home is 3.3.4

Build will not let use "#define NUM_M106_FANS 3" - will only allow value of 1.
Conditionals_post.h is only defining "#define HAS_FAN0 1" - all others are grayed out. However pins_ARMED.h has FAN_PIN, FAN1_PIN and FAN2_PIN configured as PC6, PC7 & PC8.

Other issue is that if I enable #define SPEAKER I get a timer conflict.  I don't know if it is related to the fan issue or because I'm using TMC2130 drivers. I can live without speaker though.