ktand/Armed #7: New board

milicigoran, 2021-04-15:
Sorry if its a dumb question but I picked up this board for the price and saw it could work with a prusa. I have a mk3s prusa but SS the MB and until the new one gets here it will be some time so i ordered this and it got here in a few days. But i cant find much instructions on it or other marlin files so i kind of know what to flash . Or if i can even flash ince it seems like the bootloader is locked or doesn't have one. See am just guessing flying blind here. Any help would be awesome.

ktand, 2021-04-15:
You just have to build Marlin using the PlatformIO environment (preferably using VS Code) and then you upload the firmware to the board.

milicigoran, 2021-04-17:
Is there a configuration anywhere to get a kind of jumpstart or just start from zero? Because I'm trying to use it for a prusa i3 mk3s like it says in the description that it can be a drop in replacement but i have no clue if it came preloaded with the firmware or i have to build it from scratch. really no info out there on this board. When i plug it in the red light just blinks.

milicigoran, 2021-04-17:
would this would probably work if i took the skr 1.4 turbo prusa fork and just changed the environment to this board?

milicigoran, 2021-04-18:
I tried that last night without much success. I think i had the proper
environment except i had problems with the accompanying header files .
Either they werent there or were wrong or outdated. It wouldn't let me
build with tmc 2009 but when I switched to tmc 2130 8t let me go almost to
the end of compiling before some new error messages. You dont happened to
have a copy of a compiled config.h and config.adv.h you could email so I
could just plug in to try and take a look over it and see what kind of
works and what doesn't. This is my first board that's not mainstream that
has a ton of documentation. But would love to be able to use it amd not
spend 90 bucks for einsy Rambo boards

logan8888, 2021-12-14:
I am trying to use this board but my windows 10 PC does not recognize it.....Do i need to get a driver from somewhere?